ezAuton/ezAuton #26: Suggestion: Make unit tests run inside a VM emulating the processor on the roboRIO

ritikmishra, 2018-08-06:


andrewgazelka, 2018-08-09:
This sounds like a good idea. I am guessing you are primarily doing this to make sure the roboRIO is powerful enough to run the program (CPU / memory). I wonder how this would work with automatic unit testing (i.e. Travis). Do you think it would make more sense to run the unit tests with low Xms/Xmx values?

ritikmishra, 2018-08-09:
Perhaps. I was thinking about how it would be a good idea to see if our
compilation methods are correct when it comes to getting a Java + C++ mix
running on an ARM architecture (we're going to use the JNI to reuse code
for C++ ezauton, right?)

andrewgazelka, 2018-08-09:
I definitely think we should start by using JNI. If the code efficiency becomes a major problem, we can always create our own C++ port, but we should not worry about that now.

In that case, we should definitely use some type of Virtual ARM Linux Environment. I'm just interested in seeing how this would integrate with current unit tests. Also, if it becomes too hard to do with Travis, I don't see a problem in running ARM-specific tests locally or on a VPS/dedicated server.