ezAuton/ezAuton #39: closest point doesn't consider the endpoints of lines

andrewgazelka, 2018-12-05:
[image]

ritikmishra, 2019-05-02:
I feel like this might be wanted so that the robot is closer to the implied final angle then it otherwise would be if it just drove towards the endpoint. Thoughts?

andrewgazelka, 2019-05-02:
> I feel like this might be wanted so that the robot is closer to the implied final angle then it otherwise would be if it just drove towards the endpoint. Thoughts?

I think this could be an issue for paths like 

```
B
|
|
|
|
|
+------------+
             |
             |
             |
             |
             |
+------------+
|
|
|
|
|
|
|
A
```
because the robot will just keep going straight from A->B (I think)

ritikmishra, 2019-05-02:
this only happens with the last segment

andrewgazelka, 2019-05-02:
hmmmm perhaps it is better to have it this way? I am not  sure.

ritikmishra, 2019-05-03:
I think so.

closed 2019-05-03 by ritikmishra