ezAuton/ezAuton #40: PathGenerator consider acceleration between multiple points

andrewgazelka, 2018-12-06:
Right now if the magnitude of acceleration/deceleration from point n to point n+1 is too low to complete the segment, the code will just throw an exception. We should add implementation for smoothing acceleration/deceleration between multiple waypoints, so we can easily have the fastest motion possible between _more than two_ waypoints.

andrewgazelka, 2018-12-14:
I think there is a lot of possible discussion to be had on what is the best solution. If we automatically fix impossible trajectories, that might be bad for the end-user because they might be confused how a trajectory works.

We could also use separate waypoints for defining the path and defining speed (speed waypoints would take in `<speed, distanceAlongPath>`). However, this is quite clunky as well, especially if someone changes a path after speeds are set.

On the other hand, we can make it seem like speeds set are max speeds and not speeds which will actually be obtained. Probably best to separate this from current implementation (which throws errors if a path constraint cannot be reached). Opinions?

ritikmishra, 2019-02-24:
> We should add implementation for smoothing acceleration/deceleration between multiple waypoints,

What do you mean by this? Do you mean automatically changing accel/decel values so the robot goes faster? 

> we can make it seem like speeds set are max speeds and not speeds which will actually be obtained
How would one set a desired speed, then?

andrewgazelka, 2019-02-24:
I mean if there is a waypoint 10 waypoints that you need to start decelerating right _now_ to make it to the desired speed of that waypoint, then decelerate now.