ezAuton/ezAuton #41: Look at "plugins" ability

andrewgazelka, 2018-12-13:
It might make sense for people to add plugins to the Visualizer module rather than having to modify source. Why?

- The visualizer will be the only module run directly from a client (i.e., it isn't some part of a larger jar like ezAuton Recorder/Core/WPILib...)

- it would be pretty 🔥for people to be able to add plugins to easily get more visualizers. This allows for easily adding/removing like plugins on mc servers.

andrewgazelka, 2018-12-13:
Requires #42

ritikmishra, 2018-12-13:
Shuffleboard uses plugins so it may be useful to have a peek at its code.

See:
https://github.com/wpilibsuite/shuffleboard/blob/master/app/src/main/java/edu/wpi/first/shuffleboard/app/plugin/PluginLoaderHelper.java
https://github.com/wpilibsuite/shuffleboard/blob/master/app/src/main/java/edu/wpi/first/shuffleboard/app/plugin/PluginLoader.java
https://github.com/wpilibsuite/shuffleboard/blob/master/app/src/main/java/edu/wpi/first/shuffleboard/app/plugin/PluginCache.java