ezAuton/ezAuton #45: Should Recording#update() update subrecordings?

andrewgazelka, 2018-12-13:
I vote no since this unnecessarily intertwines updating when we might not want to update all sub recordings at the same time... the main recording shouldn't know when sub recordings should update. However, is this makes everything a lot cleaner (i.e., in the visualizer) it might make more sense to do this...maybe.

ritikmishra, 2018-12-13:
The main recording *should* know when all the child recordings get updated.
A parent knows when all its children get fed, our sub-recordings aren't
some sketchy teenagers trying to party out at night without being watched
by their parents

Also, updating all sub-recordings is going to be a common thing people do
when they record multiple subsystems on their robot.

If Recording#update() didn't update subrecordings, what would it do? Would
its new behavior make it more difficult to have Recording implement
ISubRecording?

andrewgazelka, 2018-12-13:
Recording still serves as a set of ISubRecording.

> A parent knows when all its children get fed, our sub-recordings aren't
some sketchy teenagers trying to party out at night without being watched
by their parents
but what if your parent wants to force feed you and you don't want to get fed, or you aren't fed enough? If you can't get a snack yourself, that is quite bad.

> Would its new behavior make it difficult to have Recording implement ISubRecording?
No. `ISubRecording` does not extend `Updateable`

ritikmishra, 2018-12-13:
> but what if your parent wants to force feed you and you don't want to get fed, or you aren't fed enough? If you can't get a snack yourself, that is quite bad.

SequentialDataRecordings specifically can get snacks on their own if the user code updates them manually and adds them to the recording at the very end for serialization