ezAuton/ezAuton #53: Decide on versioning scheme/system

andrewgazelka, 2018-12-16:
https://github.com/ezAuton/ezAuton/issues/15#issuecomment-447620574
If we are using JitPack (mentioned by @andrewda), we probably want to start tagging commits with the current version. I think we can probably agree [Semantic Versioning](https://semver.org/) is the best route, but how should we execute this? Should we have some automatic tag system with Gradle or do it manually? I think manual is probably the best option to avoid mistakes, but I am not completely sure. What are your opinions?

ritikmishra, 2018-12-16:
`git tag` + github release system

andrewgazelka, 2018-12-21:
what do we want to do for pre-1.0.0 versions?

ritikmishra, 2018-12-21:
0.x.y

pbriggs28, 2018-12-21:
My 2 cents: 

### Gradle Managed Version Numbers
We use gradle to define our version numbers (using `group` and `version`). Publishing to maven automatically picks this up. No idea how jitpack works but I would assume it uses these too (they're pretty standard). 

### Release Types
Another point worth mentioning is the release types (`SNAPSHOT` vs `RELEASE`):

[image]

You are always working on a SNAPSHOT (ex: `1.14.0-SNAPSHOT`) and then when you create a release, you change the version to RELEASE (ex: `1.14.0-RELEASE`). The snapshot is not stable, code is changing. Releases are stable, once a version is released, it is locked in history and will never change.

Here is the git branching strategy to go with this:
[image]

### Pre 1.0
Normally a RELEASE should be fully documented, tested, etc. Prior to `1.0.0` however, it ok to break this rule. The entire code base is not considered stable until the first `1.0.0` release.

andrewgazelka, 2019-01-12:
> 0.x.y

so what if someone makes a PR and we want to increment the PATCH number but they didn't add a tag?

ritikmishra, 2019-01-12:
Merge + add tag