ezAuton/ezAuton #62: Move partly to Kotlin?

andrewgazelka, 2019-01-10:
I really love Kotlin, but I know some people who contribute to this project don't use it. Now that I have learned a bit of it, I feel so sluggish writing anything in Java... 

Kotlin is amazing because it works fine having parts of the project in Java and parts in Kotlin. Are we alright doing this? Here are some advantages of Kotlin:

- 100% interoperable Kotlin <=> Java
- wrapped reference objects (no need for AtomicObjects in lambas for reference)
- objects (Singleton shortcut)
- very well-defined immutable vs mutable collections 
- able to wrap last arg higher-order functions after a function.. i.e. `thread { println("Hello") }`
- functions not bound to classes
- no required semi-colons
- delegate methods / no need for annoying getter/setter methods (i.e., `val time get() = System.out.currentTime()`)
- `val` and `var`
- generators (and coroutines)
- operator overloading
- extension functions for bad APIs (this might be nice if we want to add extension functions to the WPILib API)
- ...

andrewgazelka, 2019-01-10:
since I am impatient, I will start in a new feature branch and delete it if people don't like it.

andrewgazelka, 2019-01-10:
also, forgot to mention... nullability is a really key aspect of Kotlin. It is very hard to get a NPE in Kotlin, but you can still use null values you add `?` after the class type. It is infinitely better than the Java `Optional<V>` class.

andrewgazelka, 2019-01-10:
and imo enforces better good practices as well (everything is by default `final`)

ritikmishra, 2019-01-10:
ok so kotlin has all these great feature

How do you want to apply them and how will that prevent people who don't know kotlin from contributing to the project?

andrewgazelka, 2019-01-10:
well, if you look at the current source code, we don't need a large percent of classes to be in Kotlin. I think a lot of people who are using special third-party libraries also have a large chance of using Kotlin. I don't think it really is going to change things up for the contributors a whole ton... especially since I am only using it for unit tests now, which people more add on to and not modify.

Probably sound completely illiterate, really tired lol

andrewgazelka, 2019-01-10:
@ritikmishra are you alright with me merging this then?

ritikmishra, 2019-01-10:
> I think a lot of people who are using special third-party libraries also have a large chance of using Kotlin

I very strongly disagree. This year, the default build system is Gradle, making it *significantly* easier to use a third party library without a lot of technical knowledge when compared to Ant.

andrewgazelka, 2019-01-10:
[image]

Proving my point :p

andrewgazelka, 2019-01-10:
look, I'll definitely move from Kotlin back to Java if people start complaining, but until that point, I think the benefits outweigh the costs.

ritikmishra, 2019-01-10:
fine.

closed 2019-01-10 by ritikmishra

AlecDusheck, 2019-01-10:
+1. The learning curve isn't huge and Kotlin looks super cool.