ezAuton/ezAuton #65: jitpack seems to be unstable

andrewgazelka, 2019-01-11:
at least from my experience, it seems like dependencies only download about half of the time for ezAuton-Core. It might just be me though.

AlecDusheck, 2019-01-11:
Can you provide log files or does it not throw anything?

andrewgazelka, 2019-01-12:
Maybe... actually noting my dependencies one was `ezAuton-Core` and one was `ezAuton`. However, I could have sworn during open room, I tried all combinations and it was still unpredictable... might just be me being stupid. I'll try again tomorrow.

andrewgazelka, 2019-01-12:
Closing as I am not (yet) having any dependency problems. Will open again if I have more problems today.

andrewgazelka, 2019-01-12:
Well...
```
 > Could not resolve com.github.ezAuton.ezAuton:ezAuton-WPILib:-b26423c914-1.
         > inconsistent module metadata found. Descriptor: com.github.ezAuton.ezAuton:ezAuton-WPILib:-SNAPSHOT Errors: bad version: expected='-b26423c914-1' found='-SNAPSHOT'
```
even though 
[image]

... confused.

andrewgazelka, 2019-01-12:
note it _does_ work with `-SNAPSHOT`... weird