ezAuton/ezAuton #66: codecov stopped working 5 months ago

andrewgazelka, 2019-01-11:
No wonder code coverage has not been increasing. For some reason it stopped 5 months ago. The last commit tested was 5b749e350ae362d4d1adad4f7071fabdb3cf1d50 and the commit after that was 48e98778cbf7f2d6e6d16e8348dafb14a8859de7. Commits are [here](https://codecov.io/gh/ezAuton/ezAuton/commits). Any idea why it suddenly stopped working?

ritikmishra, 2019-01-11:
perhaps the webhook for the repo broke or something?

andrewgazelka, 2019-01-11:
Possibly, but 
[image]

andrewgazelka, 2019-01-11:
also latest Travis-CI build:
[image]

ritikmishra, 2019-01-11:
try it manually?

andrewgazelka, 2019-01-11:
[image]
think I found the issue. We don't have a webhook enabled...

andrewgazelka, 2019-01-11:
oh nvm

andrewgazelka, 2019-01-12:
Fixed for some reason in 4cb9edcaf03ebb9c34bcd6bd288e73ad082eafd5

closed 2019-01-12 by andrewgazelka