ezAuton/ezAuton #80: ITranslationalDistanceSensor is redundant with IEncoder

ritikmishra, 2019-01-14:
as long as the answer to #69 is 
> no, IEncoder should not have getVelocity() as a method
ITranslationalDistanceSensor is identical to IEncoder

ritikmishra, 2019-01-14:
oops, doc says 

>  Like an encoder but for translational distance instead of rotations. An example of a ITranslationalDistanceSensor is an {@link EncoderWheel}.

ritikmishra, 2019-02-23:
Suppose we have `ITypicalMotor implements IVelocityMotor, IEncoder, <other non-relevant interfaces>`. Utilities like `PurePursuitMovementStrategy` expect to be able to tell the motor to go at some velocity in feet per second, *not encoder units*.R

andrewgazelka, 2019-02-23:
> o tell the motor to go at some velocity in feet per second, _not encoder units_. It does not make sense to have an `ITypicalMotor` that will accept FPS but does not output feet/fps in its IEncoder implementation

I am confused... PurePursuit would need feedback in feet not encoder units anyway... the whole signature of `ITypicalMotor` is interesting already... seems like it deserves to be multiple interfaces anyway.

ritikmishra, 2019-02-23:
The point of `ITypicalMotor` is to make the implementation of ezAuton on real robots easy. It would not be ezAuton if it was hard.

andrewgazelka, 2019-02-23:
so why doesn't `ITypicalMotor` extend `ITranslationalDistanceSensor`? I am confused... how can it be able to know velocity in fps without knowing the wheel diameter and if it knew the wheel diameter why wouldn't it be able to get the distance in feet?

ritikmishra, 2019-02-23:
Excerpt from RC2019:

```
  left = new ITypicalMotor()
        {
            public void runVelocity(double targetVelocity)
            {
                this.makeSlavesFollowMaster();
                frontLeft.set(ControlMode.Velocity, targetVelocity / Constants.Physical.DriveTrain.ENC_UNITS_TO_FPS);
            }

            public void runVoltage(double targetVoltage)
            {
                this.makeSlavesFollowMaster();
                frontLeft.set(ControlMode.PercentOutput, targetVoltage);
            }

            public double getPosition()
            {
                return (double) frontLeft.getSelectedSensorPosition(0) * Constants.Physical.DriveTrain.ENC_UNITS_TO_FEET;
            }

            public double getVelocity()
            {
                return (double) frontLeft.getSelectedSensorVelocity(0) * Constants.Physical.DriveTrain.ENC_UNITS_TO_FPS;
            }

            private void makeSlavesFollowMaster()
            {
                backLeft.follow(frontLeft);
            }
        };
```

commit b97daf1a (2019-02-24): Fixes #80
closed 2019-02-24 by andrewgazelka