ezAuton/ezAuton #85: IAction#end()

andrewgazelka, 2019-02-20:
Should `IAction#end()` only be called on peaceful ends? Even though it says this in docs, it isn't necessarily true because it is called when cancelling... `ActionGroup` with actions... if it is always peaceful wouldn't that mean it could be kept in `isFinished()` anyway (even though this is kinda jank)?

ritikmishra, 2019-02-21:
No it should be called on all ends. There should also be a method for when it is interrupted that by default calls end just in case

andrewgazelka, 2019-04-17:
Should we still even have an `Action#end()` after #106 ? fyi, [this](https://github.com/ezAuton/ezAuton/blob/155efe87f534f849daa54a89026dbe495e9b04c2/Core/src/main/java/com/github/ezauton/core/action/tangible/ActionCallable.java#L24-L35) is how actions are currently run.