ezAuton/ezAuton #91: get rid of IAction#getFinished() and IAction#onFinish()

andrewgazelka, 2019-02-22:
why? ... it is ambiguous whether `onFinish()` will be executed when interrupted or peacefully finished. The syntax is also slightly jank and might be confusing for people.

ritikmishra, 2019-02-22:
Is it used by other actions?

andrewgazelka, 2019-02-22:
> Is it used by other actions?

Action _Implementations_? No... it is currently used in unit tests and `ModernSimulatedClock`, but I am not sure it is needed there either. If we do keep it, we should definitely figure out _when_ it should be called.

andrewgazelka, 2019-04-17:
I think this whole callback syntax is kinda bad. Additionally, imo an `Action` should not worry about managing things to run on finish. I think it is better to just use an `ActionGroup` as it is more explicit. In Kotlin, we could make this an _extension function_ 
```kotlin
fun Action.after(other: Action) = ActionGroup.ofSequential(this, other)
```
we could then chain these together (which would result in nested ActionGroups, but with error propagation this is probably ok).

closed 2023-04-10 by andrewgazelka