ezAuton/ezAuton #102: Actions reusable or not

andrewgazelka, 2019-04-16:
# Should Actions be able to be Reused?
## TL;DR
If actions are reusable we will need to trust user implementation of `run(ActionRunInfo)` is pure to avoid state  conflicts. If they are not reusable, this will add more (perhaps unnecessary) restrictions to the user, but with the benefit of less confusing errors for naïve programmers. 

## Explanation
Currently the action are defined as:
[image]

There is expected to be a new instantiation for every `run(ActionRunInfo)` call. This allows us to have less state conflicts. If we reuse of actions, one process changing the state of the action being used by another process which is not expecting the state to be changed can result in issues. 

However, creating a new instance does have downsides
- WPILib Commands can be reused, so this might be unwanted variance between the standard library
- Commands cannot be reused

If actions must be reused, their properties set by their constructor should be read-only and not modified—even by `run(ActionRunInfo)`. If this is not the case, if two actions are running concurrently there might even be concurrent modification exceptions because values are attempting to be set from two places. Therefore, `run(ActionRunInfo)` would need to always be a pure function and we would need to trust that people would not implement it wrong. 

Should we allow extra freedom with the loss of having a high chance of a naïve programmer having code with confusing errors?

### Case Example of design change  with reuse  of actions
Another example of where action reuse might be nice: `ActionGroup`s. Reusing them would allow us to make a group ONCE and then easily use it in multiple places. However, note how we mentioned all properties msut be read only. This means we would not be able to have `addParallel(Action)`, `addSequential(Action)`, ... methods as if they are called when another process is already running, it will cause havoc. Instead, we would need to pass in action info into the constructor and an immutable list  of actions would be stored. Hence, we would have an `ActionGroupBuilder` for having `addSequential(Action)` and other methods. Additionally, all tracking of how many actions are left to schedule would be implemented in the  pure method, not a field in the class.

andrewgazelka, 2019-04-17:
After a very long time thinking about this, I have decided—along with help of solomon from 2898—that actions most likely should not be reused. This is due, in part, to the following reasons:
- we would need to rewrite a large portion of the `Action` framework
- there would be many breaking API changes for those using `ezAuton`
- there is not enough of a reason to convert to reusable actions as if programmed right, their behavior will be nearly identical to non reusable actions.
- getting data from `Action`s (i.e., when recording) is very tough if they are reusable due to all variables being read-only. 

If anyone else has other, differing opinions, I will re-open this issue.

closed 2019-04-17 by andrewgazelka

ritikmishra, 2019-04-17:
ok this is not pertinent to the issue but how did you make these diagrams?

andrewgazelka, 2019-04-17:
> ok this is not pertinent to the issue but how did you make these diagrams?

[IntelliJ Class Diagrams](https://www.jetbrains.com/help/idea/class-diagram.html)