ezAuton/ezAuton #108: Immutability and Pure functions with Path and PurePursuitMovementStrategy

andrewgazelka, 2019-04-19:
Currently [Paths](https://github.com/ezAuton/ezAuton/blob/d2e62895b3cf2d2a30720f6809d8e5ab0856a1d9/Core/src/main/java/com/github/ezauton/core/pathplanning/Path.java#L16) are not Immutable. For instance, they contain
```
private int segmentOnI  // the segment currently on
private ImmutableVector closestPoint // the closest point on the path the robot has been by
private ImmutableVector robotLocationClosestPoint // the robot location when getting the closest point
```
This seems like it should not belong to a path. Perhaps _path_ is just bad naming? In my opinion it couples the path-following control law to paths too much.
 
---

Let's examine the only current system which uses `Path`—[PurePursuitMovementStrategy](https://github.com/ezAuton/ezAuton/blob/master/Core/src/main/java/com/github/ezauton/core/pathplanning/purepursuit/PurePursuitMovementStrategy.java#L68-L102).
Take a look at `update(...)`. Besides updating `latest{...}` fields for the sole purpose of recordings being able to pull data, the method is pure. If we wanted to, we could write a pure, top-level function for it.

```kotlin
fun findGoal(path: Path, origin: ImmutableVector, lookahead: Double, stopTolerance: Double): Goal {
    if (stopTolerance <= 0) {
        throw IllegalArgumentException("stopTolerance must be a positive number!")
    }

    val currentSegment = path.currentSegment
    val closestPoint = currentSegment.getClosestPoint(origin)
    val currentDistance = currentSegment.absoluteDistanceOnLine(closestPoint)
    val distanceLeftSegment = currentSegment.absoluteDistanceEnd - currentDistance
    val distanceToClosestPoint = closestPoint.dist(origin)

    if (distanceLeftSegment < 0) {
        if (path.progressIfNeeded(distanceLeftSegment, distanceToClosestPoint, origin).size != 0) {
            return findGoal(path, origin, lookahead, stopTolerance) // progresses recursively until at right point
        }
    }

    val finalDistance = path.length

    val distanceLeftTotal = finalDistance - currentDistance

    if (distanceLeftTotal < stopTolerance) {
        return Goal.Finished
    }

    path.progressIfNeeded(distanceLeftSegment, distanceToClosestPoint, origin)
    val goalPoint = calculateAbsoluteGoalPoint(path, distanceLeftSegment, lookahead)
    return Goal.Waypoint(goalPoint)
}
```
Well, at least directly pure... the path state is being modified. Perhaps state should be included in `PPMovementStrategy`. 

---

Additionally, _should_ this be a top-level function? Code using it can be sloppy as it can't be in an iterator form, i.e.,

```kotlin
while(!strategy.isFinished()){ // note this assumes the first update returns a valid result
  strategy.update(...)
  val goal = strategy.goal
  // ...
}
```

vs 

```kotlin
do {
        val goal = findGoal(...)
        when (goal) {
            is Goal.Waypoint -> {
                // ... 
            }
            else -> return
        }
    } while (true)
```

andrewgazelka, 2019-04-19:
> This seems like it should not belong to a path. Perhaps _path_ is just bad naming? In my opinion it couples the path-following control law to paths too much.

I'm thinking it probably makes sense to have a "`PathTraversal`" class which would take in an immutable `Path` and behave somewhat like an iterator.

ritikmishra, 2019-04-19:
I am in agreement. Putting instance specific path related stuff in its own
class is a good idea. This way, users could store paths generated on robot
init and not have to store a waypoint builder or something like that

closed 2023-04-10 by andrewgazelka