ezAuton/ezAuton #115: should there be a delay of {period} between init() and execute() in `PeriodicAction`?

andrewgazelka, 2019-05-02:
- if there is a delay it might make more sense for things like localizers i.e., if init sets initial location and then execute looks at difference in position and assumes static dt between executes a delay might be useful

ritikmishra, 2019-05-02:
* a localizer that assumes static dt between executes is poorly designed because periodic actions can lag and run less often than you tell them to

andrewgazelka, 2019-05-02:
@ritikmishra yup. Look at https://github.com/ezAuton/ezAuton/issues/116 However, the delay might allow for cleaner code and a less-strange first `execute()` as referencing that issue effective `dt` would either be undefined or effectively 0 (if looks back at `initialize()` which I believe would be a bad value to pass in)

closed 2023-04-10 by andrewgazelka