ezAuton/ezAuton #119: figure out how units should work with ImmutableVector

andrewgazelka, 2019-05-03:


ritikmishra, 2019-05-03:
???

andrewgazelka, 2019-05-03:
> ???

See https://github.com/ezAuton/ezAuton/tree/convert-kotlin/Conversion/src/main/kotlin/com/github/ezauton/conversion 

currently `ImmutableVector` only allows `Double`s but if we are returning a 2D location it would be optimal to return an `ImmutableVector` with 2 distances, not doubles.

andrewgazelka, 2019-05-03:
perhaps we could have 
```kotlin
class Scalar private constructor(override val value: Double) : Value<Scalar> {
    override fun Number.wrap() = scalar()
    companion object {
        fun wrapScalar(number: Double) = Scalar(number)
    }
}

fun Number.scalar() = Scalar.wrapScalar(toDouble())
```

ritikmishra, 2019-05-03:
I feel like if user code does something that would involve getting an ImmutableVector consisting of units out in the end, they probably still know what units they're dealing with. Maybe we could have a wrapper class for immutable vector that attaches a type to it. I find myself averse to forcing all immutable vectors to have units attached

andrewgazelka, 2019-05-03:
Well we could have  `ImmutableVector<T>` essentially be like a list and then use extension functions to have different methods for different types. (i.e. can't multiply two `ImmutableVector<String>`)

ritikmishra, 2019-05-03:
I do not like that.

andrewgazelka, 2019-05-03:
> I do not like that.

why?

ritikmishra, 2019-05-03:
> extension functions

It goes back to this Java vs Kotlin debate. I feel that ezAuton should be easy for your typical Java team (i.e not able to write their own PP controller) to use in order to increase their technical ceiling, whereas (if I am understanding correctly) you believe that ezAuton should be easy for kotlin users and teams who would otherwise have the capability to write this code on their own, but would like to save time by borrowing our library.

Additionally, vectors have numbers. If you are going to have a vector with _not numbers_ in it, you should just have a list instead. If you do wish to do `ImmutableVector<T>`, I would urge you to do something along the lines of `ImmutableVector<T extends Double>`, although I also disagree with this, since I don't think that wrapping numbers in objects would be particularly easy to use nor necessarily very memory efficient (remember: we are running this on the _roboRIO_ and also those crappy FTC phones)

Finally, I am kind of averse to the whole idea of unit conversions. I understand that they *will* be useful for helping to prevent units errors (e.g milliseconds vs seconds), and that it follows the precedent set by `Stopwatch`, `TimedPeriodicAction`, etc in wanting to know the unit (`Duration`) for the number as well as the number itself, but I feel that, if you're not dumb and consistently use the same units, then it shouldn't be that big of a problem.

andrewgazelka, 2019-05-03:
> you're not dumb and consistently use the same units, then it shouldn't be that big of a problem.

but who consistently uses the same units? Always seconds or always milliseconds? Always inches or always feet? And how can one be sure that people in the future will not have the wrong units as well? Note for a long period of time a very confusing issue (I forgot what) in ezAuton was caused by accidentally switching milliseconds and seconds. This library would prevent this.

Additionally [extension functions can still be called from Java](https://stackoverflow.com/a/28364983/4889030).

Another option is to make `ImmutableVector<T>` an interface and have a `Double` implementation and a units implementation.

closed 2023-04-10 by andrewgazelka